vzzederzv/SE04-Nhom22.1 #7: Phân tích G-Code

vzzederzv, 2021-01-03:


doanhjut, 2021-01-03:
[image]

doanhjut, 2021-01-03:
O0024				Mã chương trình
G21G40G49G80			Chuyển về mặc định(hủy bù dao,..)
G91G28Z0				Rút dao về chuẩn tham chiếu
T2M6					Dao 2,thay dao nhanh
G90G55X0.Y0.			Dịch chuyển dao nhanh về vị trí x0,y0
G43H2Z100	Gọi chiều dài dao ở địa chỉ H2 và vừa gọi vừa chạy về vị trí z100 theo lệnh
S1200M3				Quay dao theo chiều thuận
X-60.Y-60.
Z5.
G1Z-3.F100				cắt tới vị trí z=-3 với tốc độ 100mm/phút
G41D2X-35.Y-40.F350	bù dao bên trái với khích thước dao d2 từ điểm (x,y0=(-35,-40) với tốc độ 350mm/phút
Y25.					Tới y=25
G2X-25.Y35.R10	cắt theo chiều kim đồng hồ từ vị trí(-25.35) với bán kinh 10
G1X25.				Cắt thẳng tới x=25
X35.Y25.				
Y15.
X15.
G3Y-15.R15.		cắt vòng cung ngược chiều kim đồng hồ với r=15
G1X35.				Cắt thẳng tới x=35
Y-25.		
X25.Y-35.
X-25.
G2X-35.Y-25.R10.	cắt theo chiều kim đồng hồ từ vị trí(-35.-25) với bán kinh 10
G1Y-20.			
G3X-65.R15.F800			
G40G00X-60.Y-60	Hủy bù dao ,tới vị trí (x,y)=(-60,-60)
Z5.
M5
G91G28Z0	Chuyển sang hệ tọa dộ tương đối, thu dao về về chuẩn
M30				Hết chương trình quay lại điểm đầu

closed 2021-01-03 by vzzederzv