vzzederzv/SE04-Nhom22.1 #9: Tìm hiểu và sử dụng AlphaCAM Ultimate Router

vzzederzv, 2021-01-03:


endeeayy, 2021-01-03:
update hướng dẫn sử dụng vào

Omegaduc, 2021-01-03:
File dxf: https://github.com/vzzederzv/SE04-Nhom22.1/blob/main/00154973_18mmw2440_h1220_1_Z.dxf

Omegaduc, 2021-01-03:
### **Import file DXF vào Alphacam:**
1. Chức năng Input CAD.
[image]

2. Chọn định dạng DXF.
[image]

3. Chọn file DXF.
[image]

4. File DXF mẫu.
[image]

Omegaduc, 2021-01-03:
### **Cài đặt các thông số của vật liệu:**

1. Chức năng Set Material.
[image]

2. Nhập vào thông số vật liệu.
[image]

3. Chức năng Set Geometry Z Levels.
[image]

4. Tạo độ sâu của từng nét cắt theo yêu cầu.
[image]

Omegaduc, 2021-01-03:
### **Chọn công cụ gia công và thao tác thi công:**

1. Bảng chọn công cụ gia công.
[image]

2. Chức năng Pocket Holes.
[image]

3. Thiết lập thông số của lỗ cần đục.
[image]

4. Chọn lỗ để đục.
[image]

5. Hành trình của mũi khoan.
[image]

6. Thiết lập thông số của đường cắt.
[image]

7. Chọn phần muốn cắt.
[image]

8. Hành trình của máy cắt.
[image]

9. Toàn bộ hành trình của các công cụ gia công.
[image]

Omegaduc, 2021-01-03:
### **Chạy thử giả lập:**

1. Chức năng Send to Stimulator.
[image]

2. Chế độ Giả lập.
[image]

3. Kết quả giả lập thi công.
[image]

Omegaduc, 2021-01-03:
### **Xuất file G-Code:**

1. Chức năng Output NC.
[image]

2. Chọn xuất ra File.
[image]

3. Lưu file G-Code.
[image]

Link anc: https://github.com/vzzederzv/SE04-Nhom22.1/blob/main/test2.anc

closed 2021-01-03 by Omegaduc